akospasztor/stm32-bootloader #3: How to compute the RAM_SIZE?

kiwih, 2018-02-07:
Hi,

In your bootloader.h file you have the following:
```
/* MCU RAM size, used for checking accurately whether flash contains valid application */
#if (STM32L496xx)
#define RAM_SIZE        (uint32_t)0x00040000
#elif (STM32L476xx)
#define RAM_SIZE        (uint32_t)0x00018000
#endif
```

My question is - how did you compute these values? I'm not sure how to get from 320K / 128K of RAM to those numbers.

Cheers

akospasztor, 2018-02-07:
Hi,

The RAM_SIZE define should have been named more exact - namely SRAM1_SIZE. Those numbers represent the size of the SRAM1 (used by default) in the MCU, which are 256kB for the L496xx and 96kB for the L476xx. 

In the case of the L496, you can just simply specify and use the full RAM size (320kB), because the SRAM1 and SRAM2 are continuously defined in the memory space and you don't have to switch in between the two. (Please refer to the MCU's reference manual and datasheet.)

In the case of the L476, the SRAM2 is not continuous to the SRAM1, it is placed elsewhere in the memory space, therefore you cannot simply use the full RAM size 'continuously' as it is in the L496.

Best,
Akos

kiwih, 2018-02-07:
Ok - that makes sense. Thanks!

I have another question though: the only place RAM_SIZE is used is in this function:
```
uint8_t Bootloader_CheckForApplication(void)
{
    return ( ((*(__IO uint32_t*)APP_ADDRESS) & ~(RAM_SIZE-1)) == 0x20000000 ) ? BL_OK : BL_NO_APP;
}
```

For the STM32L496xx, for example, the equation is
* APP_ADDRESS = 0x08008000
* RAM_SIZE = 0x00040000
`0x08008000 & ~(0x00040000) = 0x08000000`

Why does the address 0x08000000 evaluate to 0x20000000 when the Bootloader has loaded a program? And why such a roundabout way of calculating this address? 

Thanks heaps for your help.

akospasztor, 2018-02-07:
The CheckForApplication function uses the following principle to check accurately whether the application space contains valid application firmware: when you compile&build an application and check the output binary, the very first DWORD (first 4 bytes) of the firmware always represents the location of the stack pointer. Therefore at startup, the microcontroller knows where to initialize the stack pointer. This value is very specific, in fact the value must be within the SRAM range, which starts at 0x2000000 and the size depends on the actual microcontroller variant. For example, the L476VG has 96k SRAM1 starting from the above mentioned location, so in the CheckForApplication verifies that the first 4 bytes of the binary lies within its SRAM1 memory space. This method gives a very accurate guess whether the application space contains garbage or an actual, valid application firmware. 

Best,
Akos

kiwih, 2018-02-07:
Okay, but according to your memory diagram in the README (Figure 2)

0x08000000 is the first DWORD of the Bootloader, not the first DWORD of the Program. So how does this address end up as part of the Program?

Sorry for being confused!

akospasztor, 2018-02-07:
So the idea is to check the following: is the stack pointer of the application (located at 0x0800**8000**) lies between the RAM start address and RAM end address? It could be "translated" to the following: 
` is (0x20000000) <= (*(APP_ADDRESS)) <= (0x20000000 + RAM_SIZE - 1) ?`

However if you consider the code in the CheckForApplication function, you can verify that it does exactly the same, but this method is a creative and (more importantly) faster way than actually performing the above mentioned two comparisons.

fatalcoder524, 2021-07-20:
Hello @akospasztor . I read the above comments. I tried the same code to check if the application is present, but it always fails.
The values I'm using are:
```
#define BOOTLOADER_ADDRESS     ((uint32_t)0x08000000)
#define APPLICATION_ADDRESS     ((uint32_t)0x08008000)    /* Start user code address: ADDR_FLASH_PAGE_13 */
#define RAM_BASE                          SRAM1_BASE                  /*!< Start address of RAM */
#define RAM_SIZE                           SRAM1_SIZE_MAX           /*!< RAM size in bytes */
```

```
uint8_t Bootloader_CheckForApplication(void)
{
    return (((*(uint32_t*)APPLICATION_ADDRESS) - RAM_BASE) <= RAM_SIZE) ? BL_OK  : BL_NO_APP;
}
```

akospasztor, 2021-07-26:
Hey @fatalcoder524, 

Can you check *why* does it fail?
- What kind of MCU do you use?
- What kind of compiler / IDE do you use?
- What kind of value is present at the address of `APPLICATION_ADDRESS` (0x08008000) just before calling the `Bootloader_CheckForApplication` function?
- Are the `SRAM1_BASE` and `SRAM1_SIZE_MAX` defines correctly defined?
- Are the bootloader project options properly configured? (i.e. is the RAM start address and size properly configured so that it reflects those defines?)