akospasztor/stm32-bootloader #6: Check for app bug?

KonstantinBerezenko, 2019-04-03:
https://github.com/akospasztor/stm32-bootloader/blob/4e5e6be6ed69d30ddd05787c90dee124c830397c/Src/bootloader.c#L316

Perhaps I'm missing something but if RAM_SIZE = 0x00040000 then (RAM_SIZE - 1) = 0x0003ffff, then inverted it equals 0xfffc0000.   Then 0x20040000 & 0xfffc0000 = 0x20040000 which is not equal to 0x20000000.  This assumes that the first 4 bytes of the app point to the end of the RAM space which I think it does at least on my app.

Rotule666, 2019-04-03:
I am using this successfuly on multiple applications :
`return ( ((*(__IO uint32_t*)APP_ADDRESS) - RAM_SIZE) == 0x20000000 ) ? BL_OK : BL_NO_APP;`

I updated it in my implementation of the bootloader (in which I use CAN) but never raised the issue, I remember I had troubles with it.

KonstantinBerezenko, 2019-04-03:
That implementation makes sense to me.

pycan, 2019-09-13:
Hi Akos,

Thanks for you reply.

I could not find the new implementation of this section in the develop branch, they look similar (maybe not pushed to the remote repo yet?)

In either case, I have updated this function in my code to what is shown below

` return (((*(__IO uint32_t*)APP_ADDRESS & (~RAM_SIZE-1)) <= (uint32_t)(0x20000000+RAM_SIZE)) ? BL_OK : BL_NO_APP);`

akospasztor, 2019-09-16:
> Thank you Akos for your reply.
> 
> I could not find the new implementation of this section in the develop branch, they look similar (maybe not pushed to the remote repo yet?)

Yeah, sorry, you are absolutely right. I haven't pushed my local repo yet, as it has currently many new branches with new and experimental stuff :D As my free/time allows in the next couple of days, I will sort out my locals and push the changes. As many curious people - like you :) - want to see new stuff (which i'm always happy for), I just don't want to publish anything even to the develop branch without basic working functionality.

kubasaw, 2020-07-22:
Hi, 
First of all, I want to thank you @akospasztor for publishing well commented code, what helped me a lot to understand more or less, code execution workflow on the STM.

I have some issue with porting Bootloader to STM32F446. On this MCU, there is 128KB (0x0020 0000) of SRAM available, starting at: 0x2000 0000, so the last byte of RAM is placed at: 0x201F FFFF.
For every application code which I have generated by arm-none-eabi-gcc the first DWORD is set to 0x2020 0000 what causing false evaluation of condition (0x0020 0000<0x0020 0000) in CheckForApplication.

The condition proposed by @Rotule666: 
```return ( ((*(__IO uint32_t*)APP_ADDRESS) - RAM_SIZE) == 0x20000000 ) ? BL_OK : BL_NO_APP;```
seems correct for me in contrast to:
```return (((*(uint32_t*)APP_ADDRESS) - RAM_BASE) < RAM_SIZE) ? BL_OK : BL_NO_APP;```
in the v1.1.0 code.

Where I make a mistake in my way of thinking?

Rotule666, 2020-07-22:
@kubasaw 

`return ( ((*(__IO uint32_t*)APP_ADDRESS) - RAM_SIZE) == 0x20000000 ) ? BL_OK : BL_NO_APP;`

this always worked out for me except for the H7

A good trick to check what should be in there is to flash your application and check at APP_ADDRESS usinf st-link utility.

akospasztor, 2020-07-22:
Hi guys, thanks for the comments.

@kubasaw: I just looked into the issue you mentioned, and you're right. The location of the stack pointer is set to the upper boundary of the RAM, seemingly 4byte outside of the RAM space (0x0020 0000). However, when there is a `push` instruction, the core first decreases the value of the stack pointer and then writes to the (already decreased) location. Therefore, having an initial stack pointer at 0x00200000 is valid.

I would like to point out that @Rotule666 's solution is correct, as long as the stack pointer is always configured at the end of the RAM region. However, be careful, as this is not the case if there is a custom placement, or another compiler is used (e.g. iar, armcc, etc.), and this can lead to problems. Therefore, I would advise to use the following check (`<=` instead of `<`):

`return (((*(uint32_t*)APP_ADDRESS) - RAM_BASE) <= RAM_SIZE) ? BL_OK : BL_NO_APP;`

I will incorporate this change in the next release.

kubasaw, 2020-07-22:
Hi,
Thank you @akospasztor for your confirmation, but one more question arises. We want to check if stack pointer is set to point at RAM region, is `(((*(uint32_t*)APP_ADDRESS) - RAM_BASE) <= RAM_SIZE)` a correct comparison?
When we change comparison operator from `<` to `<=` it virtually increases RAM_SIZE by 1, so in my opinion, in extremal case, when stack pointer is set to RAM_BASE, after `push` it will point at other (not valid) region.
In my opinion, `(((*(uint32_t*)APP_ADDRESS) - RAM_BASE-1) < RAM_SIZE)` will be better solution ;)

akospasztor, 2020-07-22:
The stack pointer location cannot be set to the `RAM_BASE`, because the stack grows "downwards", i.e. decreases. Therefore, it must be greater (at least 4 bytes) than the `RAM_BASE`.

Secondly, if a microcontroller has the `RAM_BASE` mapped to 0x0000 0000 (instead of 0x2000 0000), then your proposed solution would cause an overflow.

kubasaw, 2020-07-22:
You are right, I have also made mistake. I completely agree.

Rotule666, 2020-07-22:
@akospasztor I agree 100%, this is therefore why I had to change it for my project for H7 where I have put the Stack in DTCM RAM

good job :)

PR #12 (2019-09-20): Add fix for CheckForApplication and introduce semantic versioning
This merge request attempts to fix #6 and introduces semantic versioning. More information: https://semver.org. 

Additionally, the PLL_P divider is adjusted.
PR #16 (2019-10-26): Merge v1.1.0 release into master
This merge request introduces the new stm32-bootloader version. This release contains a complete repository overhaul, a new implementation for the stm32l496 discovery board and much more. Please refer to the changelog and documentation for detailed description of changes and features.

This merge request closes related issues: it closes #14 closes #11 and closes #6.
PR #38 (2020-07-22): Fix a boundary case in CheckForApplication function
This PR fixes the boundary in `Bootloader_CheckForApplication` function when the stack pointer location is set to the upper bound of the RAM.

Closes #6
closed 2020-07-22 by akospasztor