akospasztor/stm32-bootloader #7: Why do you have to reset these struct members? I have not seen this in other bootloaders, but it works for me.

jakeammons, 2019-05-31:
https://github.com/akospasztor/stm32-bootloader/blob/4e5e6be6ed69d30ddd05787c90dee124c830397c/Src/bootloader.c#L328

jakeammons, 2019-05-31:
Why do you have to reset these struct members? I have not seen this in other bootloaders, but it works for me.

akospasztor, 2019-06-03:
Because it's really important to reset every peripheral and their registers to their initial, reset-state. After the jump, the application firmware itself has no knowledge (and should not have) whether it was launched from a bootloader. It assumes it was launched after power-up or reset, thus it expects every peripheral to be in reset state.

As an example referring to your question: when you don't reset the systick registers, thus you don't stop the systick operation, it can cause some nasty issues when there is a tick interrupt just after the moment of jump. Another issue would be when for example the application firmware does not use systick at all, therefore there is no interrupt handler defined for it but the interrupt is coming - since it hasn't been disabled before the jump.

jakeammons, 2019-06-03:
That all makes sense. Thanks so much for the help!