akospasztor/stm32-bootloader #13: Error : Jump to application

Rajkumar181, 2019-10-11:
Hi,
   I am trying this example for my STM32F429zi, and I can erase and flash the memory but I am not able to run the application code, please can you suggest to me where I am doing wrong.

I am attaching my code. if any other details are required please let me know.

Thanks
[BootTest.zip](https://github.com/akospasztor/stm32-bootloader/files/3716693/BootTest.zip)

akospasztor, 2019-10-11:
Hey there, 

first of all, let me kindly tell you that I've been writing and developing these projects for my own education / hobby / entertainment / etc., and I've decided to publish them because I believe it might be useful and beneficial for many other people. I always try to provide sufficient and detailed documentation, alongside with detailed comments in the source code, so that people can easily continue and customize my projects and examples to their taste. I always welcome any kind of feedback, feature request, etc., and I always try to answer to all of the questions and issues within a reasonable time. I am happy to help people and provide guidelines to the best of my knowledge to solve their problems. However, please understand that I don't have any resources, nor time to do any code review or write code for another person. 

With that being said, I would recommend to perform the followings: 
1. Check if the flashed application code is written properly into the MCU flash. Is the application binary written properly, to the right location?
2. In order to perform a successful jump, the following operations need to be performed:
    - reset all peripherals to their reset state
    - disable systick (if running)
    - make sure there isn't any interrupt pending to be served
    - set the new vector table location
    - set the stack pointer
    - jump
3. Try to debug and step the jump, and observe what happens exactly, what does the MCU. Do you jump to the correct location?

Let me know if this helps.

Best,
Akos

Rajkumar181, 2019-10-21:
Thanks for your support. 
I followed your suggestion but I am not able to jump the correct location. Sorry, I am turbling to you.

my MCU is - STM32F429zi

This my jump code

___disable_irq();
uint32_t  JumpAddress = (*(__IO uint32_t*)(APP_ADDRESS + 4));
 pFunction Jump = (pFunction)JumpAddress;
 HAL_RCC_DeInit();
 HAL_DeInit();
 SysTick->CTRL = 0;
 SysTick->LOAD = 0;
 SysTick->VAL  = 0;

 SCB->VTOR = APP_ADDRESS;
 __set_MSP(*(__IO uint32_t*)(APP_ADDRESS));
 Jump();

Rajkumar181, 2019-10-21:
When I tried in a different way, it is jumping to the correct location but still, the problem is same.

__disable_irq();
			uint32_t  JumpAddress = ((APP_ADDRESS + 4));
		    pFunction Jump = (pFunction)JumpAddress;

		    HAL_RCC_DeInit();
		    HAL_DeInit();

		    SysTick->CTRL = 0;
		    SysTick->LOAD = 0;
		    SysTick->VAL  = 0;

		#if (SET_VECTOR_TABLE)
		    SCB->VTOR = APP_ADDRESS;
		#endif

		    __set_MSP((APP_ADDRESS));
		    Jump();

Why we are doing pointer cast?

akospasztor, 2019-10-26:
When you set the stack pointer, you have to set it to the value defined at the very first 32-bit dword of your application. If you write `__set_MSP((APP_ADDRESS));` then you set the stack pointer to the address of your application and not to the value pointed by this address.

Therefore, the correct solution would be: `__set_MSP(*(__IO uint32_t*)APP_ADDRESS);`

closed 2020-10-22 by stale[bot]