akospasztor/stm32-bootloader #54: Support for STM32F4 series

ashishsk, 2021-09-15:
Hi, I am trying to adapt this library for STM32F401 MCU and it seems like there are a ton of configurations that are different. Could you suggest what's the best way to port this implementation to F4?

akospasztor, 2021-09-15:
Hi Ashish,

It depends on what kind of hardware setup you have. Usually porting the bootloader-specific HAL commands from L4 family to F4 family should be pretty much straightforward if I remember correctly. The STM32 comes with tons of examples for F4 microcontrollers, and they are also available on GitHub (I'm not sure how up-t-date this is): https://github.com/STMicroelectronics/STM32CubeF4.

For instance, you can take a look at this example how to use the flash-related HAL commands on F401: https://github.com/STMicroelectronics/STM32CubeF4/tree/master/Projects/STM32F401-Discovery/Examples/FLASH/FLASH_EraseProgram

If you also would like to port the SD-Card-related (or other) functionalities, then you need to dig a little bit deeper and understand what exactly are the differences between the peripherals of L4 and F4 microcontrollers. For this, I would recommend to check out the reference manual of the F401 (https://www.st.com/resource/en/reference_manual/rm0368-stm32f401xbc-and-stm32f401xde-advanced-armbased-32bit-mcus-stmicroelectronics.pdf). 

I hope this helps. 
Akos

ashishsk, 2021-09-15:
Hi Akos, 

thanks for your inputs. I have a basic setup: uSD card breakout board communicating with STM32F401-Nucleo vis SPI. I used the FATfs driver to read the SD card, which has a text file and a binary file which needs to be flashed on the controller. I managed to read the text file fine but I did not manage to copy the contents of the binary to the flash memory. In the example for Flash on F401, I see there is an implementation for reading the contents of the flash, erase, etc but I am a bit lost on how to copy the contents from the SD card to the flash in the first place. Since this is my first project using the STM32 or the HAL, the examples of the flash seem a bit overwhelming for a beginner. I was of the opinion, the easiest approach to learn was to have a working example such as yours(L series) on my board (F series) and hence the question of porting. I would be glad to get your ideas for porting.

Regards, 
Ashish

closed 2021-09-15 by ashishsk