akospasztor/stm32-bootloader #55: How to build the firmware image

joa77, 2021-11-08:
Hello,
thank you for your great bootloader code, which is the first code, which runs perfectly for me and is easy understandable.
But I have one question. How can I build the firmware update image?
CubeIDE gives me an Intel hex file, .ELF file or an binary file which is almost 140MB large.
Thank you for your help

akospasztor, 2021-11-08:
Hi joa77,

just to clarify: are you referring to the application image, that you would like to flash into the microcontroller with the bootloader? Or the bootloader code itself?

Generally, you need to generate a simple binary output (usually a .bin file) for the application image to be able to flash it with the bootloader (as it currently does not support .hex or any other formats than binary). 

Nevertheless, the 140 MB is huge. I have no idea how you could have managed to have an application binary with 140MB. Can you please share some info about your project and how you build it?

Best,
Akos

joa77, 2021-11-08:
Thank you for the quick reply.
I am talking about the actual app image here.
The bootloader is only a few kb in size.
Too large bin files seem to be a general problem with ST controllers. Google gives a lot of results after a short query.
For example: https://community.st.com/s/question/0D50X00009XkYIP/bin-file-generated-by-gcc-too-large

It seems to be because Flash and RAM do not form a contiguous memory address area. So the space in between is filled with zero.

By adjusting the linker script, I have now managed to have only the flash area in the .bin file and thus only a few kb in size.

Do I now have to fill the .bin file with zeros at the end, so that it can fill the complete flash, so that the CRC is correct?

Translated with www.DeepL.com/Translator (free version)

akospasztor, 2021-11-09:
Yes, you have to adjust your linker configuration to generate the correct binary.

If you want to utilize the CRC feature, you have to fill up the remaining space (except the actual 4-byte CRC value) with `0xFF`. If you want to use another CRC method or other CRC configuration, you can edit the verification function here:

https://github.com/akospasztor/stm32-bootloader/blob/7472c7b439d1921ecf9b835bc8e85b53c0549aa9/lib/stm32-bootloader/bootloader.c#L353

closed 2022-01-16 by stale[bot]