akospasztor/stm32-bootloader #57: Not an issue - a question - SCB->VTOR issue with F103?

arkhipenko, 2022-02-27:
Hi,

This is a shot in the dark and a cry for help.
I am writing a simple bootloader for STM32 F103 series using Arduino IDE. 
I figured all the settings and am pretty sure I have set the linker loader script and VTOR table right.
I am confident of that because when I load the target firmware to 0x80080000 address with the cube programmer it works fine and VTOR is set correctly. 

However, when I jump to the firmware from the bootloader, the VTOR is not set to 0x8008000! It remains 0x8000000. No idea why!
It is such a simple code and yet, something does not work. 
Have you encountered any issues with F103 line?

Thank you!

```
void boot_jump2Address(const uint32_t address)
{
  uint32_t appStack;
  pFunction appEntry;
 
//  Serial.print("boot_jump2Address=0x"); Serial.println((uint32_t)address, HEX);

  // get the application stack pointer (1st entry in the app vector table)
  appStack = (uint32_t)*((__IO uint32_t*)address);
//  Serial.print("appStack=0x"); Serial.println((uint32_t)appStack, HEX);
 
  // Get the app entry point (2nd entry in the app vector table
  appEntry = (pFunction)*(__IO uint32_t*)(address + 4);
//  Serial.print("appEntry=0x"); Serial.println((uint32_t)appEntry, HEX);

  delay(1000);

  
  HAL_RCC_DeInit();
  HAL_DeInit();
 
  SysTick->CTRL = 0;
  SysTick->LOAD = 0;
  SysTick->VAL  = 0;
 
  // Reconfigure vector table offset to match the app location
  delay(100);
  
  __disable_irq();
  SCB->VTOR = 0x8008000;
  __DSB();
  __enable_irq();

  __set_MSP(appStack); // Set app stack pointer

  appEntry(); // Start the app
 
  while (1); // never reached
}
```

akospasztor, 2022-02-28:
Hi arkhipenko,

My guess is that the startup code of the application firmware explicitly sets the `VTOR` back to the starting address of the flash. What kind of startup code do you use? I remember that the cube-generated startup code in `system_stm32l4xx.c` explicitly sets the `VTOR` in the `SystemInit()` function.

Cheers,
Akos

akospasztor, 2022-03-01:
Hi arkhipenko,

I don't exactly know what `pinMode()` does in your `setup()` (nor do I know what `setup()` does actually 😅), but does that maybe set/change the boot pins or boot configuration? If yes, then your problem was actually that you changed the boot mode of the MCU, which you don't need :) 

Anyway, I'm glad that you could resolve the problem!

Cheers,
Akos