akospasztor/stm32-bootloader #63: Is it a bug in function Bootloader_Erase()?

KiraVerSace, 2022-11-24:
OK, here is your soure code, if I use a MCU only 256KB like stm32l476rc, all the flash in the bank1 from page1 to 128(every page is 2K), so the NbrOfPages must be < FLASH_PAGE_NBPERBANK(256), but in your program, it will skill the ` if(NbrOfPages > FLASH_PAGE_NBPERBANK)`.

And the program will not erase the BANK1, cause it will be skipped.

Maybe i was wrong, Thank you for your contribution, it help me a lot, it write very clearly and easy reading.
```
uint8_t Bootloader_Erase(void)
{
    uint32_t NbrOfPages = 0;
    uint32_t PageError  = 0;
    FLASH_EraseInitTypeDef pEraseInit;
    HAL_StatusTypeDef status = HAL_OK;

    HAL_FLASH_Unlock();

    /* Get the number of pages to erase */
    NbrOfPages = (FLASH_BASE + FLASH_SIZE - APP_ADDRESS) / FLASH_PAGE_SIZE;

    if(NbrOfPages > FLASH_PAGE_NBPERBANK)
    {
        pEraseInit.Banks     = FLASH_BANK_1;
        pEraseInit.NbPages   = NbrOfPages % FLASH_PAGE_NBPERBANK;
        pEraseInit.Page      = FLASH_PAGE_NBPERBANK - pEraseInit.NbPages;
        pEraseInit.TypeErase = FLASH_TYPEERASE_PAGES;
        status               = HAL_FLASHEx_Erase(&pEraseInit, &PageError);

        NbrOfPages = FLASH_PAGE_NBPERBANK;
    }

    if(status == HAL_OK)
    {
        pEraseInit.Banks     = FLASH_BANK_2;
        pEraseInit.NbPages   = NbrOfPages;
        pEraseInit.Page      = FLASH_PAGE_NBPERBANK - pEraseInit.NbPages;
        pEraseInit.TypeErase = FLASH_TYPEERASE_PAGES;
        status               = HAL_FLASHEx_Erase(&pEraseInit, &PageError);
    }

    HAL_FLASH_Lock();

    return (status == HAL_OK) ? BL_OK : BL_ERASE_ERROR;
}
```

KiraVerSace, 2022-11-24:
The blow is my program inherit from you.

```
bool bootLoaderEraseFlash(void)
{
    /**
     * @brief 定义FLASH每个Bank有多少Pages
     *        STM32L4系列分为两个Bank,每个Bank=256Pages 每个Page=2K
     */
    uint32_t numberOfPages = 0;
    uint32_t pageError = 0;

    FLASH_EraseInitTypeDef flashEraseInit;
    HAL_StatusTypeDef halStatus = HAL_OK;

    HAL_FLASH_Unlock();

    numberOfPages = (FLASH_BASE+FLASH_SIZE-USR_APP_ADDRESS)/FLASH_PAGE_SIZE;

    flashEraseInit.Banks     = FLASH_BANK_1;                                        // 擦除指定的Bank 目前用的L476RC只有一个Bank
    flashEraseInit.NbPages   = numberOfPages % (FLASH_BANK_SIZE/FLASH_PAGE_SIZE);   // 一次性擦除多少页
    flashEraseInit.Page      = (USR_APP_ADDRESS-FLASH_BASE)/FLASH_PAGE_SIZE;        // 擦除的起始页,从除去BootLoader开始
    flashEraseInit.TypeErase = FLASH_TYPEERASE_PAGES;                               // 擦除类型是“Page Erase”

    halStatus = HAL_FLASHEx_Erase(&flashEraseInit, &pageError);

    if ((halStatus == HAL_OK)&&(numberOfPages > (FLASH_BANK_SIZE/FLASH_PAGE_SIZE)))
    {
        flashEraseInit.Banks     = FLASH_BANK_2;
        flashEraseInit.NbPages   = numberOfPages - flashEraseInit.NbPages;          // 减去上次擦除的Pages得到剩余的
        flashEraseInit.Page      = USR_APP_ADDRESS-(FLASH_BASE+FLASH_BANK_SIZE)/FLASH_PAGE_SIZE;
        flashEraseInit.TypeErase = FLASH_TYPEERASE_PAGES;

        halStatus = HAL_FLASHEx_Erase(&flashEraseInit, &pageError);
    }
    HAL_FLASH_Lock();

    if (halStatus == HAL_OK)
    {
        logInfo("IAP Erase the flash [0x%08X-0x%08X] successfully!", USR_APP_ADDRESS, FLASH_SIZE);

        return true;
    }
    else
    {
        logError("IAP Erase the flash error!");
        return false;
    }
}
```

akospasztor, 2022-11-30:
Hi @KiraVerSace,

If you use a microcontroller that has flash banks with number of pages other than 256, then you need to change that in the `bootloader.h` file.

I quickly double-checked the datasheet of the [STM32L476RC](https://www.st.com/resource/en/datasheet/stm32l476rc.pdf), and it states explicitly that each flash bank contains 256 pages. See page 19:

> STM32L476xx devices feature up to 1 Mbyte of embedded Flash memory available for storing programs and data. The Flash memory is divided into two banks allowing read-while-write operations. [...] Each bank contains 256 pages of 2 Kbyte.

However, digging into the reference manual of the microcontroller, this does not seem to be true for devices with 512KB flash size... See page 98 here: https://www.st.com/resource/en/reference_manual/rm0351-stm32l47xxx-stm32l48xxx-stm32l49xxx-and-stm32l4axxx-advanced-armbased-32bit-mcus-stmicroelectronics.pdf

So yes, you need to adapt the configuration in the `bootloader.h` so that the `FLASH_PAGE_NBPERBANK` reflects the correct number of flash pages in a bank.

closed 2023-02-11 by stale[bot]